CamelCaseName/HUB75nano #3: Write Big Char

fcabrera23, 2020-12-03:
Hi, I was checking your library and its not possible to write a big string, like for example 10:30 in the whole LED display. Do you have any function for that?

CamelCaseName, 2020-12-03:
Hi, unfortunately I only have the small font (5x3) implemented..

You could set every pixel the number should use manually/in a function, but apart from that, I have nothing for this. Sorry. But I am currently working on a new project, which will bring some improvement to this one also

CamelCaseName, 2020-12-03:
you can take a look at my Word clock example, which displays characters on the panel, but very small ones.

fcabrera23, 2020-12-03:
Hi, thanks for your quick response. I would be willing to pay for the deployment, do you think you can make it? I need this asap

fcabrera23, 2020-12-03:
My idea es showing a text like  " 23.5 | 76%" which is the temp and humidity. Thanks

CamelCaseName, 2020-12-03:
i added a function with a scaling factor for the font it currently uses, so you can have big chars, but still blocky. just call it stylized and done

fcabrera23, 2020-12-04:
Hi, @CamelCaseName  thanks for all your help. I have just tried and getting some errors with the display. Im attaching prictures

[image]

[image]

CamelCaseName, 2020-12-04:
ok, this looks weird. i saw something similar with my tries yesterday, and i got it solved by reducing ram usage. can you please check in the "panel.h" whether  "#define BIG" is commented or not. if you comment this line, you can only set the first 8 colors(0-7), but it might fix this problem. with the BIG statement set, the buffer uses two bits per pixel, which comes out to a total of nearly 95% ram usage, which doesn't always work. with only one bit per pixel, you have about 800bytes of ram to work with. it might also be the wiring,  there is nothing more I can think of right now.

if you take a look at the "word clock example", there should be some info on the ram issues.

i could also try to reduce variables used, but it is getting kind of complicated

CamelCaseName, 2020-12-04:
the one thing that is extra weird is the fact, that the letters are along the short side, which I have no idea on how they got there. 

i might look into a vertical mode as well, but I can not guarantee anything

fcabrera23, 2020-12-06:
Thanks.. I was using horizontal mode, just took the picture in a vertical mode. I will try your solution and let you know tomorrow.

CamelCaseName, 2020-12-07:
i tried your code with both big and small buffer, and it had no effect.
[image]
the number looks normal, even the small red bar at the top is kind of normal, i dont know where these bleeds come from, or how to get rid of them. the only thing i think might cause your problem are wrongly connected row select lines.

i included your code in the exampe, so you can try also

have fun

closed 2021-03-23 by CamelCaseName