CamelCaseName/HUB75nano #4: Letters repeating

aherrera1997, 2021-01-14:
Hello, I've been testing your code and I'm having an issue. I uploaded your library to the Arduino Nano and with this script

`
#include "Panel.h"

//create an instance of the panel
Panel panel(32, 64);
char out[81];
uint8_t length = 0;
uint8_t oldlength;
uint8_t fcolor = 3;
uint8_t bgcolor = 2;
char t;

void setup() {
  panel.createBufferBG(panel.BLACK);//background
 
}

void loop() {

  panel.drawBigChar(3,8,'A',panel.RED,3);
  panel.drawBigChar(14,8,'|',panel.GREEN,1);
  panel.displayBuffer(); //makes the buffer visible and the leds all blinky blinky
}`

I get this output

[image]

I should get a red 'A' and a green '|'. What might be the problem?

CamelCaseName, 2021-01-14:
hi, i tried out your code with no modifications, and it works flawlessly on my panel.
[image]

I suggest you take a look at your wiring, because the duplication looks like maybe a address line is swapped or so.
The buffer size has no effect on the output, i checked with both big and small buffer.
i'd like to help you more, but i sadly cannot reproduce this problem.

if someone can reproduce this effect, please message me or create a new issue/reply to this one.

i would suggest you to use another library, but as far as i am aware, mine is the only one which works on a nano.

have fun with your panel.
kind regards, 

Leonhard

aherrera1997, 2021-01-20:
Which arduino are you using for this?

CamelCaseName, 2021-01-20:
i am using a cheap arduino nano clone with the original atmel ATmega328p precooked with the "old bootloader"^.
Pinout and board layout is the exact same, so shouldnt be a difference there.
i cannot find the name of my exact clone.

aherrera1997, 2021-01-20:
It seems we are using the same copy. And the panel?

CamelCaseName, 2021-01-20:
i bought some cheap panel on ebay, using the standard hub75 protocol, so no weird row selection with mux oder identifier sequence.
[image]

so yeah, i am also kind of out of ideas

aherrera1997, 2021-01-20:
We have different panels it seems, maybe thats the problem. Mine is P5(2727).

CamelCaseName, 2021-01-20:
well the p is just the pitch (distance between middle of leds), so no factor. but iirc the 2121 and 2727 are different types of led, maybe thats the problem

aherrera1997, 2021-01-25:
Have you seen something about some panels having 2 GND and a E input? I read about it and now I think it could be the issue

CamelCaseName, 2021-01-25:
well the E pin is for the 5th bit in the row selection to make use of 64 rows, but I have no experience so far with these Panels.

aherrera1997, 2021-02-04:
I changed the panel for a P4 2121 and now it's working correctly, thanks for the support.  I have another question now, is there an easy way to change the design of the numbers?

CamelCaseName, 2021-02-04:
im glad you got it working now. changing the font is not a trivial task, i got this super tiny font somewhere from the internet, together with an unwrapping function. if you go looking for tiny fonts, like maybe 8 by 4 or 10 by 5 pixels, i may implement them if the license allows for this kind of use. or you can figure out how the unwrapping part works, and compile your own font to this. i dont really have the time right now to do this myself.

i hope i could help you with this info

regards, leonhard

closed 2021-03-23 by CamelCaseName