CamelCaseName/HUB75nano #7: Can this library work with a 1/8 scan rate display?

Abdelalihub, 2021-04-15:
Hi leonhard,

Thanks a lot for this library and the examples.
I have a question for you :
I have a 64x32 rgb display with a 1/8 scan rate, can this library drive this display too? if not, is it possible to have it working with some minimal updates, like updating some parameters on the library?

Kind regards,
Abdelali

CamelCaseName, 2021-04-15:
uh, i haven't tried it yet. From just thinking about it i would say no, because i have no idea how 32(16 because two color pins) rows are addressed with 3 bits. but in theory it shouldn't be too hard to get it running for 1/8th scan instead of 1/16th scan.

CamelCaseName, 2021-04-15:
you could try just hooking it up without the 4th cable, and see what happens.

Abdelalihub, 2021-04-15:
Hello,
Thanks a lot for the fast reply.
I can fill the whole display with a color using this function `fillScreenColor()`, but it just displays the RGB colors (RED, GREEN, BLUE) correctly, others don't work, in most times I get a RED color.

With the `createBufferBG()`, it display the colors correctly but just on 4 rows (1st line in each 8 rows). The rows in between are OFF.
Showing a character just doesn't work. In most times I get 3 separated dots at the center of the display.

Another question please :
Is this line :
`PORTC = (k + (k > 7) * 8) + (k > 15); //sets all rows `
The equivalent of  this function : 
```
void Panel::selectLine(uint8_t c) {//selects one of the 16 lines, 0 based
   switch (c) {
       case B0000:
           digitalWrite(RA, HIGH);
           digitalWrite(RB, HIGH);
           digitalWrite(RC, HIGH);
           digitalWrite(RD, HIGH);
           digitalWrite(RA, LOW);
           digitalWrite(RB, LOW);
           digitalWrite(RC, LOW);
           digitalWrite(RD, LOW);
       break;
       case B0001:
           digitalWrite(RA, HIGH);
       break;
       case B0010:
           digitalWrite(RB, HIGH);
       break;
       case B0011:
           digitalWrite(RA, HIGH);
           digitalWrite(RB, HIGH);
       break;
       case B0100:
           digitalWrite(RC, HIGH);
       break;
       case B0101:
           digitalWrite(RA, HIGH);
           digitalWrite(RC, HIGH);
       break;
       case B0110:
           digitalWrite(RB, HIGH);
           digitalWrite(RC, HIGH);
       break;
       case B0111:
           digitalWrite(RA, HIGH);
           digitalWrite(RB, HIGH);
           digitalWrite(RC, HIGH);
       break;
       case B1000:
           digitalWrite(RD, HIGH);
       break;
       case B1001:
           digitalWrite(RA, HIGH);
           digitalWrite(RD, HIGH);
       break;
       case B1010:
           digitalWrite(RB, HIGH);
           digitalWrite(RD, HIGH);
       break;
       case B1011:
           digitalWrite(RA, HIGH);
           digitalWrite(RB, HIGH);
           digitalWrite(RD, HIGH);
       break;
       case B1100:
           digitalWrite(RC, HIGH);
           digitalWrite(RD, HIGH);
       break;
       case B1101:
           digitalWrite(RA, HIGH);
           digitalWrite(RC, HIGH);
           digitalWrite(RD, HIGH);
       break;
       case B1110:
           digitalWrite(RB, HIGH);
           digitalWrite(RC, HIGH);
           digitalWrite(RD, HIGH);
       break;
       case B1111:
           digitalWrite(RA, HIGH);
           digitalWrite(RB, HIGH);
           digitalWrite(RC, HIGH);
           digitalWrite(RD, HIGH);
       break;
   }
}
```

Thanks again.

Kind regards,
Abdelali

CamelCaseName, 2021-04-15:
hi,
fillscreencolor() just pushes the selected color as fast as it can, without much "thinking", that's probably why it works.

CreateBufferBG() holds a buffer for the same amount of pixels as your panel has, but they are output in a wrong way. for your panel each row has to be clocked 128 times, not just 64 times. plus, you only have to go through 8 rows, not 16

yes, the are identical in function. for iterating through less rows, it should be sufficient to just replace the k iterator from 16 to 8 in the for loop. then the 4th pin will never be set.

CamelCaseName, 2021-04-15:
if you could fork the project, i can help you edit if you want

Abdelalihub, 2021-04-15:
Hi,

ha, I wouldn't ask. 

I didn't know that the scan rate would make that much difference.

Thanks again.

Yes, please. I'll really appreciate if you could help me on that.

I will fork the project and invite you.

CamelCaseName, 2021-04-15:
i just skimmed through the code (haven't worked on it in like half a year or more), and the biggest part of it should work without changing a lot. First of all, you have to declare the panel with 128x16 pixels, this should fix some things. 

then i need to know which rows are misplaced when displaying a buffer. I guess that the 0th row is continued in the 8th, and the 16th row is continued in the 24th, just guessing. 
Fixing this should be as easy as adding some offsets. i hope.

Abdelalihub, 2021-04-15:
Okay, thank you.

I will test with the 128x16 pixels and see.

Abdelalihub, 2021-04-15:
[image]
Still same with 128x16. With the `createBufferBG(GREEN) `.

CamelCaseName, 2021-04-15:
huh, interesting. in that case, it could be a lot of things, like the order in which to send data, and latch the input, and setting the OE pin, so on and so on. 
I've had the same problem too, and my solution was trying to find out the needed order of instructions for the panel.

i sadly think i can't help you with this problem any more...

you must excuse me, but this exceeds my capabilities.

i wish you a lot of fun and ingenuity.

CamelCaseName, 2021-04-15:
i'll close this issue then, if you need any more help, just contact me. maybe i can help.

closed 2021-04-15 by CamelCaseName

Abdelalihub, 2021-04-15:
Okay no problem.
I will try to make it work. 
Thanks for your time.