CamelCaseName/HUB75nano #12: Had a query for Panel Chaining?

anandrajgupta, 2021-08-12:
Hi Mr. Seidel,
First of all Thank you very much for this greatly optimized library.
I wanted to add 2 panels daisy chained together by connecting the output port of first panel to the input port of 2nd panel. Does this library support that type of Chaining?
Would I have to change something to accomplish that?
Any help would be highly appreciated.
Thank you!

CamelCaseName, 2021-08-12:
Hi, i'm glad you like my project.

anandrajgupta, 2021-08-12:
Wow, Thank you for the quick reply.

I am using Uno, will that be powerful enough? (The IC is same though i guess, ATMega328).

Actually I'm trying to chain 2 P10 panels of resolution 16x32 each, they have scan rate of 1/4 and they also operate on HUB75.
The overall resolution that i want to get is 32x32, will that be possible by replacing all 64 by 128?

Yes i saw the Adafruit library but it has very limited support for different types of panels, mostly runs on Adafruit Panels only (8 Scan).
And Yes i saw hzeller library also using Raspberry Pi.

Wow more optimizations would be really helpful for a lot of DIY enthusiasts, yeah it's very difficult to take out time from our regular work.

Yes i saw your write up mentioning Adafruit's point that Uno is limited to 16x32 only, I was really impressed by that article :D
I don't understand  what is that logic inside Adafruit's library which makes Uno completely blank when i run a 32x32 example in it while a 16x32 example run perfectly well. The same 32x32 example runs very well when I use Arduino Mega, what could be the limiting parameter when switching from 16x32 example to 32x32 example which is being filled by Arduino Mega but Uno fails to fullfill, i see only SRAM and Flash size difference between Uno and Mega which does not seem to be a limiting factor in Uno, i just don't understand why!

If you suggest ATMega328 can't run 2 x (16x32) then please suggest me the next best version of an Atmega controller which can do that, Thank you very much...

CamelCaseName, 2021-08-12:
Hi again, 
the Uno should be powerful enough to drive 32x32 LEDs.
by chaining two panels i thought you wanted two 32x64 panels, which probably will not work. but with two 16x32 it should.

I see two ways how you can do this:
1. just chain them how you think you would do it, so connect the output of one to the input of the other. you will then have a "16x64" panel, which my library can drive. just use it normally, but the "lower half" of the standard 32x64 is empty. This should work. 
2. Or you take the "D" address line and address the panels in parallel. So for rows 0 to 7 (0000 - 0111) you address the upper panel the normal way, and for rows 8 to 15 (1000 to 1111) you address the lower panel, essentially using the D address line as a switch. but for that you would need an AND and a NAND gate for every line and every panel. This would be easier to address, as it would actually represent a 32x32 panel, not 16x64. this is basically how it is node internally in a 32x64 panel. I suggest you try out option 1 first.

As for the difference between the mega and uno, I think its also the number of gpio pins, but I could be wrong there (probably am).

anandrajgupta, 2021-09-27:
Tried, I was able to lit few pixels of my 1/4 scan 16×32 RGB panel but couldn't chain two panels with desired results. Probably this library doesn't support Chaining.
Thanks!

CamelCaseName, 2021-09-27:
Ok, thanks for trying anyways and sharing your results! I'll close this Issue then and wish you good luck with your project.

closed 2021-09-27 by CamelCaseName