CamelCaseName/HUB75nano #14: Unable to Upload to Nano 33 IoT

augustluhrs, 2022-05-25:
hello, just trying to run the basic color-example on a nano 33 iot for a 16x32 HUB75 panel. Haven't changed anything in the script or library and receiving the following errors. I went in and changed all the `B0*` and `B1*` to `0b*` to remove that error, but no luck with the rest. Any idea what I'm doing wrong or what needs to be updated in the library to work with a nano 33 iot? Thanks!

```
Arduino: 1.8.13 (Windows 10), Board: "Arduino NANO 33 IoT"

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp: In constructor 'Panel::Panel(uint8_t, uint8_t)':

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:221:1: error: uninitialized const member in 'const uint8_t [96][2] {aka const unsigned char [96][2]}' [-fpermissive]

 Panel::Panel(uint8_t height,uint8_t width){

 ^~~~~

In file included from C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:1:0:

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel75.h:58:36: note: 'const uint8_t Panel::font4x6 [96][2]' should be initialized

         const uint8_t font4x6[96][2];

                                    ^

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp: In function 'void cnvColor(uint16_t, uint8_t*, uint8_t*, uint8_t*)':

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:435:31: warning: 'B00001111' is deprecated: use 0b00001111 instead [-Wdeprecated-declarations]

             *rt = (c >> 11) & B00001111;//in binary: RRRRRGGGGGGBBBBB >> 11 = 00000000000RRRRR

                               ^~~~~~~~~

In file included from C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/api/ArduinoAPI.h:26:0,

                 from C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/arduino.h:23,

                 from C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel75.h:46,

                 from C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:1:

C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/api/Binary.h:131:3: note: declared here

   B00001111 DEPRECATED(0b00001111) = 15,

   ^~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:436:30: warning: 'B00001111' is deprecated: use 0b00001111 instead [-Wdeprecated-declarations]

             *gt = (c >> 5) & B00001111; //in binary RRRRRGGGGGGBBBBB >> 5 = RRRRRGGGGGG, & B00111111(63) = 0000000000GGGGGG

                              ^~~~~~~~~

In file included from C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/api/ArduinoAPI.h:26:0,

                 from C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/arduino.h:23,

                 from C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel75.h:46,

                 from C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:1:

C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/api/Binary.h:131:3: note: declared here

   B00001111 DEPRECATED(0b00001111) = 15,

   ^~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:437:23: warning: 'B00001111' is deprecated: use 0b00001111 instead [-Wdeprecated-declarations]

             *bt = c & B00001111;  //in binary RRRRRGGGGGGBBBBB & 31(B00011111) = 00000000000BBBBB

                       ^~~~~~~~~

In file included from C:\Users\jaluh\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.8.13\cores\arduino/api/ArduinoAPI.h:26:0,
[... 925 lines cut ...]
   ^~~~~~~~~

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:618:9: error: 'PORTD' was not declared in this scope

         PORTD |= B00100000;

         ^~~~~

C:\Program Files (x86)\Arduino\libraries\HUB75nano-master\src\Panel.cpp:618:9: note: suggested alternative: 'PORTC'

         PORTD |= B00100000;

         ^~~~~

         PORTC

```
truncated errors because post too long

```
exit status 1

Error compiling for board Arduino NANO 33 IoT.

```

augustluhrs, 2022-05-25:
the panel75.h name is just because it was conflicting with another panel.h library I had, so I renamed it to avoid the error.

CamelCaseName, 2022-06-01:
Ok, haven't worked on the library in quite some time but I'll try my best. from what I see, it doesn't like the port names. basically i use a technique where you have 4 bytes that represent the state of the gpio pin, probably the newer Arduino libraries use a different way of naming/setting these. i would suggest you look into how you directly access port registers on your hardware, and use the (probably) new way to set them.

CamelCaseName, 2022-06-01:
> cannot bind non-const lvalue reference of type '_EPortType&' to an rvalue of type '_EPortType'

this is the error that is the issue, shown on all port accesses. one thing it could also be is that the binary operation syntax changed, but i am not sure why and how.

closed 2022-10-19 by CamelCaseName