CamelCaseName/HUB75nano #15: Library not working on Arduino Nano

RomketBoi, 2023-05-04:
When trying to #include <HUB75nano.h> with the board and port set to "Arduino Nano RP2040 COM3 Serial Port (USB)", Arduino IDE gives an error: 

`error: #error "This library only supports the Arduino nano and Uno, so the atm368p with 2kb sram, 1kb eeprom and 32kb flash. For other chips/boards, please see the internet or try to adapt this library here, but no guarantees"`

even though I am using an Arduino Nano

CamelCaseName, 2023-05-04:
Hi, the nano you have is the "Arduino Nano Connect", which has the same form factor as the original nano but a very different CPU and architecture. As stated in the error message, this library is only compatible with the ATmega328p, which is used in the original nano and the uno.

I'm sorry for this inconvenience, but so far you'll have to look elsewhere for a library for the RP2040.

CamelCaseName, 2023-05-15:
I just got an arduino nano connect with an RP2040 in the mail, will update this library when I find the time. it will probably need a level shifter though, as the rp2040 is a 3.3v cpu but the panels usually expect 5v on their channels

closed 2023-05-15 by CamelCaseName