CamelCaseName/HUB75nano #17: Problems with my nano and 64x32 panel

impressgroupsc, 2023-11-08:
Hi, I'm trying to use this library and I've started with the simplest, color-example.ino, but I can't get it to show anything on the screen.
My connections between HUB75 and Arduino change in some pin with yours, these are the ones that I use and that I have changed in the HUB75NANO.h

// MY current pin numbers
#define RA A0 // register selector a
#define RB A1 // register selector b
#define RC A2 // register selector c
#define RD 12 // register selector d
#define RE 18 // register selector e
#define RF 2 // network first byte
#define GF 3 // green first byte
#define BF 4 // blue first byte
#define RS 5 // network second byte
#define GS 6 // green second byte
#define BS 7 // blue second byte
#define CLK 8 // clock signal
#define LAT A3 // data latch
#define OE 9 // output enable

RE is not conected because my panel only have A,B,C,D

Setting this configuration, the panel remains black. If I modify the CLK to pin 9 and remove pin 12 from RD, at least rows 1-8 and 16-24 light up but the rest remain black. I have checked the connections in case there were any errors but they are the ones I put in at the beginning.

I have even followed the steps mentioned in

https://projecthub.arduino.cc/CamelCaseName/running-a-32x64-rgb-led-panel-with-only-an-arduino-nano-092d16

and with those steps I have managed to illuminate all the pixels but of course, only with one color.

My panel is a P5 64x32 1/16

Any suggestions on where I can get the error?

Thanks.

CamelCaseName, 2023-11-08:
Hi, 
the issue is that the code expects all row pins to be in order (so A0->A3 or 10->13)

I'll add a workaround and publish a new version that can work around this limitation

impressgroupsc, 2023-11-08:
Ok, thanks.
I changed D pin from 12 to A4 and now example fills entire screen, but now i'm trying to draw a circle or a text and nothing happens.
I'll wait for update, thanks.

CamelCaseName, 2023-11-08:
strange, with the correct pins as you have them you should be able to draw as well

have you tried the drawing example?

that should be more or less the best place to see how to use the functions provided

impressgroupsc, 2023-11-08:
Finally I have connected the pins according to your diagram and I have gotten everything to work perfectly. The examples have worked correctly except for the text in which I got letters fliped horizontally.
However I cannot use this configuration as I need some pins for other functions such as A4 and A5 for SDA and SCL.

CamelCaseName, 2023-11-08:
oh right, I'll also add a toggle for horizontal and vertical flips :)

and the feature for other pins for row and color setting is already in the works, probably gonna come tomorrow

CamelCaseName, 2023-11-09:
the new version 2.1.1. should work for you now, you can just define the pins how you want them before including the library

flipping is coming in the next update

impressgroupsc, 2023-11-09:
I try your new code with this pinount:

#define RA 14 // row selector a
#define RB 15 // row selector b
#define RC 16 // row selector c
#define RD 18 // row selector d
#define RE 19 // row selector e
#define RF 2 // red first byte
#define GF 3 // green first byte
#define BF 4 // blue first byte
#define RS 5 // red second byte
#define GS 6 // green second byte
#define BS 7 // blue second byte
#define CLK 8 // clock signal
#define LAT 17 // data latch
#define OE 9 // output enable

and the simplest, color-example.ino, but I can't get it to show anything on the screen.

Sorry.

CamelCaseName, 2023-11-09:
works for me with that pinout, its just the example that is broken

the big font, drawing, serial output and flash examples work

impressgroupsc, 2023-11-09:
ok, yes, i've tested with these examples and works great.

Thanks.

CamelCaseName, 2023-11-10:
if you encounter any other issues, feel free to open a new issue here

closed 2023-11-10 by CamelCaseName