CamelCaseName/HUB75nano #20: Where is support of HUB75E type panels (icnd2153, stp1612pw05, FM6124C or similar chips)  as declared in Readme?

board707, 2024-02-07:
Hi
It would be interesting to see your code for supporting panels with more complex drivers - icnd2153, FM6124C ... etc
However, the files in the `src/output/hub75e` directory are mostly empty. Where is this code located?

CamelCaseName, 2024-02-07:
Hi, the merging is not yet complete and therefore the code is still only in this file [here]
(https://github.com/CamelCaseName/HUB75Enano/blob/3210eed2a566dce8080ce56cb6d9437bcf576d4c/src/HUB75Enano.h#L1125)

I also have a gist on the setup I used [here](https://gist.github.com/CamelCaseName/59bd58b002521df8daf4efca69d01e2e)

CamelCaseName, 2024-02-07:
I had it running on my panel with fm6124c 128x64 p1.875 and on a panel of someone else with icnd2153, so they should be compatible in that sense.

the seller of the others panel stated this line encoding is called 5958(which apparently should work with most 1.875p panels), maybe that can be of help in your search.

CamelCaseName, 2024-02-07:
Basically you should be able to drive the panels without initialization with default settings.

one thing that I find very weird is that at least on my panel, each input is buffered once and therefore synchronized to the clock line, even the so called "asynchronous" clock which totally breaks that idea of glck being asynchronous

board707, 2024-02-07:
> the seller of the others panel stated this line encoding is called 5958(which apparently should work with most p1.875 panels)

Sorry, but this phrase suggests that you don't know what you're talking about.
Multiplexers (line encoding) and drivers are two independent chips that can be installed in any combination. So it’s completely unclear why you’re talking about 5958 multiplexer when I ask you about the driver.
In addition, the number "p1.875" only means the distance between pixels in millimeters and is in no way related to the type of driver on the matrix. Matrices p1.875  can have any driver.

CamelCaseName, 2024-02-07:
To be honest I haven't once searched 5958 so I did indeed not know what it was. I only relayed the sellers info that was forwarded to me. And yes I know of the pitch, but most if not all flex panels unter p2 use these pwm chips, opposed to the normal led row drivers on larger panels.

row indexing is no issue, so please disregard my statement then.

CamelCaseName, 2024-02-07:
But to make my point, I used the stp1612pw05 datasheet to get fm6124c chips on my panel running which also worked on this linked panel by a friend with icnd2153 chips

row indexing was done with a simple shift register you load 1 into and shift it down with 32 pulses. 

every! single input pin is buffered in 2 octal bus receivers and latched/unlatched with the normal clk l. also glck which should run asynchronously.... see the discussion in hzellers rpi RGB matrix library for more discussions on if that makes sense. Though I think you were in there as well :)

this is all info I have. I may suggest you also take a look at the data sheets linked in the thread where some guy found icnd2153 register configs in a Chinese forum

board707, 2024-02-07:
> So it's the seller who doesn't know what he's talking about

If you understand that this is nonsense, why are you repeating it?  :) Since you work with these panels, you should understand the matter.

Outputting colors data to a line of pixels is done by the driver, and switching of such lines is done by a multiplexer. These microcircuits operate completely independently and can be used in any combination.
In order to work correctly with the panel, you need to know the both types of chips.

>  see the discussion in hzellers rpi RGB matrix library for more discussions

You can find me there reporting on my work with similar drivers :)

CamelCaseName, 2024-02-07:
> If you understand that this is nonsense, why are you repeating it? :) Since you work with these panels, you should understand the matter

I didn't know the name of the multiplexing used...

board707, 2024-02-07:
> Did you get your panels to work 

Yes,  I got to work the panels that I have. It were with drivers FM6353 and 6363. I didn't have a panels with icdn2153.

Unfortunately, there is very little information on such drivers, so I am interested in all projects whose authors tried to work with such panels.

CamelCaseName, 2024-02-07:
Nice!

if you ever have those around you now have a starting point :)

board707, 2024-02-07:
So you too, if you will interested in FM63x3 drivers :) You can find the link in the discussion in hzellers library

Thanks, see you later

CamelCaseName, 2024-02-07:
If i have more info I'll let the discussion in hzellers lib know

closed 2024-02-07 by CamelCaseName