CamelCaseName/HUB75nano #21: Set Font to Display

tahafathi2013, 2024-03-08:
I used your library and examples but the font is not my favorite.
How can I use my favorite font or at least the international English font?

CamelCaseName, 2024-03-08:
eh it's rather complicated there is no easy way
you would need to replace the method that draws the characters with your own and your own font encoding

CamelCaseName, 2024-03-08:
but it should have all letters, no?
you can find the font I used linked in the font files

tahafathi2013, 2024-03-08:
Ok 
How set or change a font on code?

CamelCaseName, 2024-03-08:
https://github.com/CamelCaseName/HUB75nano/blob/master/src/fonts/3x5font.h is where the font code is currently located

this here is where i got it from, also has an explanation on how it works
https://hackaday.io/project/6309-vga-graphics-over-spi-and-serial-vgatonic/log/20759-a-tiny-4x6-pixel-font-that-will-fit-on-almost-any-microcontroller-license-mit#header

basically you have a row of bits and just go through each row of a letter and turn the leds on or off

CamelCaseName, 2024-03-08:
i cant really give more info than this, i plan on making the adafruit fonts compatible sometime this year

tahafathi2013, 2024-03-08:
Please put the next update in such a way that you can define the R1 R2 G1 G2 B1 B2 pins manually, as well as the ability to determine the number of panels and the desired font whose file is attached to the project.

CamelCaseName, 2024-03-08:
> R1 R2 G1 G2 B1 B2
you can already

but it is not advised as it will be way slower

> as well as the ability to determine the number of panels
so far panel linking is not supported, but if you just have them in a row it works if you define the size correctly