CamelCaseName/HUB75nano #22: row offset issues when testing on Arduino R4 Minima

ludicrousdisplay, 2024-03-22:
Hello, thanks for making this available. I am testing this out (version 2.3.1) with an Arduino R4 Minima and Waveshare RGB LED matrix (https://www.waveshare.com/wiki/RGB-Matrix-P2.5-64x32). 

When I run the drawing example the diagonal line is broken into multiple line segments, with some segments up or down 4 pixels from where they should be relative to the starting segment. I am wondering if you can suggest a starting point for me to track down the issue.

CamelCaseName, 2024-03-22:
Hi,

this could very well be a wiring issue where the C and D pins are flipped

I'll also try my r4 minima later today to test as well

ludicrousdisplay, 2024-03-22:
Cool thanks it works now! I swapped the C and D pins as you suggested. 

C now connects to 13 and D now connects to 10. Although I don't know if the Waveshare matrix has their C and D pins flipped and mislabeled or the R4 minima spec is changed. If you test on yours that might provide an answer.

CamelCaseName, 2024-03-22:
Glad to hear you got it fixed that easily!

I'll go ahead and close this issue then, but this weekend I planned on doing some more work on the library so investigating this bit firs perfectly as a start :)

Once I determine the reason I'll update the readme and instructions. Thanks for reporting the issue!

closed 2024-03-22 by CamelCaseName

CamelCaseName, 2024-03-22:
It was literally a typo in the readme... fixed now