CamelCaseName/HUB75nano #24: arduino uno R3 doesn't work

imane-jym, 2024-06-29:
I download code and run example. error blew:
[image]

I review code, I found src\boards\board_methods.h.
boards\uno code file is empty~~~

The documentation mentions redefining pins, There seems to be other code that needs to be added as well.
Thanks in advance.

CamelCaseName, 2024-06-29:
oh nice that you have a R3, 
I can try and add the definitions the next days and you can test them

imane-jym, 2024-06-29:
yes， I have it. Looking forward to your addition. Thanks very much.

adrianLach, 2024-11-30:
Hey, I copied nano.h, nano_methods.h and the nano_pin_helpers.h into src/boards/uno and updated the #defines for the uno board and renamed them respectively. Got it runnig, however I now got weird artifacts when running the flashing example.

I haven't used the LED panel before so I am not quite sure, if it is a problem with my panel.
Propably just the NANO code not working for the UNO.

Do you have an idea what the issue could be?

[image]

CamelCaseName, 2024-12-01:
on the image it seems your psu is running into its current limit?
these panels easly take 5.1v at 2A, so about 8 to 10W

CamelCaseName, 2024-12-01:
also i quickly checked and copying the pins and methods from the nano should work :D

CamelCaseName, 2025-03-12:
@adrianLach did you figure it out?

adrianLach, 2025-03-12:
I haven't looked into it again. I can give it a try and let you know.

adrianLach, 2025-03-12:
https://github.com/user-attachments/assets/3320b3ee-c1d8-45eb-8521-f25ea68e8413
[image]

Yeah seems good now. It really must have been the current limit. Completely overlooked that.

The scrolling-example how ever looks kinda off. The letters of the words get mixed up. "Hello, Horld"

But I can confirm now, that copying the header files for the nano works with the uno.

CamelCaseName, 2025-03-13:
good to know, I'll change up the library to use the nano code for the uno r3 when compiling.

regarding the scrolling, W and H just look pretty similar, with W the middle pixel is a bit lower

it's a 5x3 pixel font, therefor it's a bit ugly.

thanks for testing and I'll close this issue after I publish the new version with r3 support

CamelCaseName, 2025-04-07:
the way the half is split here shows the address pins work correctly. but r2 g2 and B2 don't.

Luking977, 2025-04-08:
Yes, Iam only Running the scrolling-example