CamelCaseName/HUB75nano #26: Broken flash example on UNO

markushevpro, 2025-01-02:
Hi there. I think there is something wrong with the flash buffer implementation - red & green addressing broken in the example. Other examples are ok. 

I've failed to fix myself, because can't understand the structure of the stored bitmap =)
Arduino UNO Rev3 + P3 64x32 HUB75

[image]

CamelCaseName, 2025-01-02:
right, I updated the bit structure and the tool to generate the images but didn't update the example...

I'll update them soon. but the linked tool to generate them is updated for the last library version

CamelCaseName, 2025-01-02:
should be fixed with 2.4.1., could you please check? It should also be updated soon in the arduino library registry

markushevpro, 2025-01-03:
Yes, works like a charm. thank you. Btw, could you please provide description of the bit structure? I would like to write self convertion tool from jpg/png

CamelCaseName, 2025-01-03:
sure, its 6 bits per byte, pixel after pixel row by row

the values themselves are constructed as follows
https://github.com/CamelCaseName/BMP2HUB75/blob/601b6b28f309afa6e24be4c3330c8d3ed67953f7/Program.cs#L60
so bit 7 and 6 are unused, bit 5 through 3 is rgb for the lower half and 2 through 0 is rgb upper half

it allows for 4 bits per pixel, just ordered by significance.
so we have 2048 values for the mmsb, then 2048 for the msb, 2048 for the lsb and 2048 for the llsb.

pretty simple structure, i just laid out all bytes in the order they would be sent to the panel

closed 2025-01-03 by markushevpro